avr-rust/blink #1: Can't compile blink example

eflukx, 2017-10-07:
Just compiled the avr-rust toolchain (2hr process!). Linked avr-rust and set as default toolchain using rustup.

When trying to build the blink example I get the following output:

```
$ XARGO_RUST_SRC=~/code/avr-rust/src/  xargo build --target avr-atmega328p --verbose
+ "rustc" "--print" "sysroot"
+ "rustc" "--print" "target-list"
+ "cargo" "build" "--target" "avr-atmega328p" "--verbose"
   Compiling arduino v0.1.0
     Running `rustc --crate-name arduino /root/.cargo/registry/src/github.com-1ecc6299db9ec823/arduino-0.1.0/src/lib.rs --crate-type lib --emit=dep-info,link -C debuginfo=2 -C metadata=4e899856c1d3bf9a -C extra-filename=-4e899856c1d3bf9a --out-dir /home/piggy/code/avr-rust-blink/target/avr-atmega328p/debug/deps --target avr-atmega328p -L dependency=/home/piggy/code/avr-rust-blink/target/avr-atmega328p/debug/deps -L dependency=/home/piggy/code/avr-rust-blink/target/debug/deps --cap-lints allow --sysroot /root/.xargo`
   Compiling blink v0.1.0 (file:///home/piggy/code/avr-rust-blink)
     Running `rustc --crate-name blink src/main.rs --crate-type bin --emit=dep-info,link -C debuginfo=2 -C metadata=ef6a619ddb2bb88f -C extra-filename=-ef6a619ddb2bb88f --out-dir /home/piggy/code/avr-rust-blink/target/avr-atmega328p/debug/deps --target avr-atmega328p -L dependency=/home/piggy/code/avr-rust-blink/target/avr-atmega328p/debug/deps -L dependency=/home/piggy/code/avr-rust-blink/target/debug/deps --extern arduino=/home/piggy/code/avr-rust-blink/target/avr-atmega328p/debug/deps/libarduino-4e899856c1d3bf9a.rlib --sysroot /root/.xargo`
rustc: /home/piggy/code/avr-rust/src/llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp:8141: void llvm::SelectionDAGISel::LowerArguments(const llvm::Function&): Assertion `InVals.size() == Ins.size() && "LowerFormalArguments didn't emit the correct number of values!"' failed.
error: Could not compile `blink`.

Caused by:
  process didn't exit successfully: `rustc --crate-name blink src/main.rs --crate-type bin --emit=dep-info,link -C debuginfo=2 -C metadata=ef6a619ddb2bb88f -C extra-filename=-ef6a619ddb2bb88f --out-dir /home/piggy/code/avr-rust-blink/target/avr-atmega328p/debug/deps --target avr-atmega328p -L dependency=/home/piggy/code/avr-rust-blink/target/avr-atmega328p/debug/deps -L dependency=/home/piggy/code/avr-rust-blink/target/debug/deps --extern arduino=/home/piggy/code/avr-rust-blink/target/avr-atmega328p/debug/deps/libarduino-4e899856c1d3bf9a.rlib --sysroot /root/.xargo` (exit code: 1)
```

Any ideas?

shepmaster, 2017-10-07:
Can you try building in release mode (`xargo build ... --release`)?

eflukx, 2017-10-07:
Yes, that results in a successful build (my first avr-rust build ever actually.. 😉).
@shepmaster Thanks for the quick response!

Could you elaborate what goes wrong when building in debug (i.e. non-release) mode?

closed 2017-10-07 by eflukx

shepmaster, 2017-10-07:
> Could you elaborate what goes wrong when building in debug (i.e. non-release) mode?

We aren't fully sure! 😅 We have avr-rust/rust#57 open for it. My gut feeling is that there's some extra register pressure in debug mode that trips some edge case of LLVM.