avr-rust/blink #7: Cannot build newer version

massimiliano-mantione, 2018-11-15:
Which version of avr-rust should be used to build this?
And, most importantly, which version of libcore?

I have just built avr-rust from HEAD (commit 512f3ae0cb18e9bb975a8d9fb4c73cd93d49ccbb) and I cannot find a matching libcore.

shepmaster, 2018-11-16:
> I cannot find a matching libcore.

You can now use the libcore supplied with the compiler.

massimiliano-mantione, 2018-11-16:
> You can now use the libcore supplied with the compiler.

If I do, I get this error:

```
[massi@matrix blink]$ rustup run avr-toolchain xargo build --target avr-atmega328p --release
   Compiling core v0.0.0 (/home/massi/bots/rust/src/libcore)
rustc: /home/massi/bots/rust/src/llvm/lib/CodeGen/MachineFunction.cpp:186: void llvm::MachineFunction::init(): Assertion `Target.isCompatibleDataLayout(getDataLayout()) && "Can't create a MachineFunction using a Module with a " "Target-incompatible DataLayout attached\n"' failed.
error: Could not compile `core`.

To learn more, run the command again with --verbose.
error: `"cargo" "build" "--release" "--manifest-path" "/tmp/xargo.AtVJDQPJyrz6/Cargo.toml" "--target" "avr-atmega328p" "-p" "core"` failed with exit code: Some(101)
stack backtrace:
   0:     0x56086006ba0c - backtrace::backtrace::trace::ha819e434b39788d7
   1:     0x56086006ac82 - <backtrace::capture::Backtrace as core::default::Default>::default::hac314a56ba24f0e3
   2:     0x56086006acf8 - backtrace::capture::Backtrace::new::h61ad3639ef0f6a7c
   3:     0x56086006a08c - error_chain::make_backtrace::hb7c02aa4aa411e9d
   4:     0x56086006a114 - <error_chain::State as core::default::Default>::default::haec1062cab32e7a0
   5:     0x560860035b2b - <std::process::Command as xargo::extensions::CommandExt>::run::h81734ae6af1d7954
   6:     0x56086002e7f1 - xargo::sysroot::build::h0613a3e516175ec6
   7:     0x56086003226e - xargo::sysroot::update::h4769e5b82aea10cb
   8:     0x56086003f8f6 - xargo::main::ha8b86cd3abf6ad67
   9:     0x560860025eb2 - std::rt::lang_start::{{closure}}::h06e4410a28c0af66
  10:     0x56086007ac52 - std::rt::lang_start_internal::{{closure}}::h322e7fb3a16ff1af
                        at libstd/rt.rs:59
                         - std::panicking::try::do_call::h572dc6f90deb1a4d
                        at libstd/panicking.rs:310
  11:     0x56086009e9a9 - __rust_maybe_catch_panic
                        at libpanic_unwind/lib.rs:103
  12:     0x560860087495 - std::panicking::try::h1d50996ce033dd53
                        at libstd/panicking.rs:289
                         - std::panic::catch_unwind::hb74f3af6b62bbba3
                        at libstd/panic.rs:392
                         - std::rt::lang_start_internal::hce01021c3c1cf20d
                        at libstd/rt.rs:58
  13:     0x5608600403b3 - main
  14:     0x7f7189f9d11a - __libc_start_main
  15:     0x560860017029 - _start
  16:                0x0 - <unknown>
[massi@matrix blink]$ 
```

When I sow the error I presumed it was because the compiler still cannot handle libcore, this is why I was looking for a reduced one.

I even tried to manually patch the one that comes with the compiler (well, a copy of it, of course) to make it similar to the "reduced" libcore, but in the end I always got this error.

Maybe the included `Xargo.toml` should stop pointing to the avr-rust "reduced" libcore.
Which is the exact command that must be run to build this sample?

If it is relevant, I also have another issue: the target json file (`avr-atmega328p.json`) is missing the `target-c-int-width` property. Without it, xargo refuses to run the build.

I added it, as `"target-c-int-width": "16"`, is this correct?

And anyway, which is the exact avr-rust git hash that works for you?
Or maybe this issue should be opened inside the avr-rust rust project?

Thanks!

massimiliano-mantione, 2018-11-16:
As additional data, this is my target file:

```
{
  "llvm-target": "avr-unknown-unknown",
  "cpu": "atmega328p",
  "target-endian": "little",
  "target-pointer-width": "16",
  "target-c-int-width": "16",
  "os": "unknown",
  "target-env": "",
  "target-vendor": "unknown",
  "arch": "avr",
  "data-layout": "e-p:16:8-i8:8-i16:8-i32:8-i64:8-f32:8-f64:8-n8-a:8",
  "executables": true,
  "linker": "avr-gcc",
  "linker-flavor": "gcc",
  "pre-link-args": {
    "gcc": [
      "-Os",
      "-mmcu=atmega328p"
    ]
  },
  "exe-suffix": ".elf",
  "post-link-args": {
    "gcc": [
      "-Wl,--gc-sections"
    ]
  },
  "no-default-libraries": false
}
```

dylanmckay, 2018-11-16:
> Maybe the included Xargo.toml should stop pointing to the avr-rust "reduced" libcore.

It looks like it was removed [13 days ago](https://github.com/avr-rust/blink/commit/93fdb9db39f19a59360e52edbd0f7f315cca7c3f#diff-73320d2d67728eb6e932b7cd575a2ec7)

> I added it, as "target-c-int-width": "16", is this correct?

Perfect!

dylanmckay, 2018-11-16:
> I think that those lines should not harm any regular case and they do allow compilation on avr.
Maybe I should issue a pull request to rust-avr?

Hm, I think upstream Rust has been receptive of a few of these 16-bit only fixes. @shepmaster may be better able to comment though.

shepmaster, 2018-11-16:
> this is my target file:

[this was also updated](https://github.com/avr-rust/blink/blob/master/avr-atmega328p.json)

shepmaster, 2018-11-16:
> The problem with the `avr-unknown-unknown` target built into Rust

Well, and we haven't set all the settings, either. AFAIK, AVR doesn't support position independence, for example. 

> if JSON files could be derived from target specifications already implemented in the compiler

You can start a path to this:

```
$ rustc -Z unstable-options --print target-spec-json --target avr-unknown-unknown
```

Coupled with something like `jq`, the transforms could be automated.

stappersg, 2021-02-04:
Closing for two reasons
* The long silence
*  AVRust being merged with upstream Rust since July 2020

In case I'm too eager with this close, please do more then clicking on `re-open`, such as creating a fresh issue.

closed 2021-02-04 by stappersg