avr-rust/blink #11: Compilation fails at before compiling libcore

1inguini, 2019-02-22:
I followed instructions for `avr-rust` and `blink` compilation. But I am getting this error:
~~~
➤ rustup run avr-toolchain xargo build --target avr-atmega328p --release
warning: Patch `rustc-std-workspace-core v1.0.0 (/home/linguini/Desktop/rust/rust-avr/src/tools/rustc-std-workspace-core)` was not used in the crate graph.
Check that the patched package version and available features are compatible
with the dependency requirements. If the patch has a different version from
what is locked in the Cargo.lock file, run `cargo update` to use the new
version. This may also occur with an optional dependency that is not enabled.
error: failed to run `rustc` to learn about target-specific information

Caused by:
  process didn't exit successfully: `rustc - --crate-name ___ --print=file-names --sysroot /home/linguini/.xargo -Z force-unstable-if-unmarked --target avr-atmega328p --crate-type bin --crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro` (exit code: 1)
--- stderr
error: Error loading target specification: Could not find specification for target "avr-atmega328p"
  |
  = help: Use `--print target-list` for a list of built-in targets

error: `"cargo" "build" "--release" "--manifest-path" "/tmp/xargo.GqdRz0LJNIMX/Cargo.toml" "--target" "avr-atmega328p" "-p" "core"` failed with exit code: Some(101)
stack backtrace:
   0: error_chain::make_backtrace::h90af62d5c4a6fbd6 (0x55fde99c316b)
   1: <error_chain::State as core::default::Default>::default::h23f817a122243ac9 (0x55fde99c3235)
   2: <std::process::Command as xargo::extensions::CommandExt>::run::h01346a6b4f35f543 (0x55fde9976d4b)
   3: xargo::sysroot::build::h240260972f2e09ed (0x55fde997e541)
   4: xargo::sysroot::update::h94fd82f52808b493 (0x55fde998265d)
   5: xargo::main::ha282b52dabbcaa07 (0x55fde996fc7a)
   6: std::rt::lang_start::{{closure}}::h6c099b348d99c7de (0x55fde99738f2)
   7: std::rt::lang_start_internal::{{closure}}::h53f6e74893a27d59 (0x55fde99ee162)
             at src/libstd/rt.rs:59
      std::panicking::try::do_call::h8945e3e0fe63365a
             at src/libstd/panicking.rs:307
   8: __rust_maybe_catch_panic (0x55fde99fb1a9)
             at src/libpanic_unwind/lib.rs:102
   9: std::panicking::try::hee49dda4da841cb1 (0x55fde99ee058)
             at src/libstd/panicking.rs:286
  10: std::panic::catch_unwind::h89d0b17e1bb724c9 (0x55fde99e1665)
             at src/libstd/panic.rs:398
      std::rt::lang_start_internal::hcbec2a0475a1169a
             at src/libstd/rt.rs:58
  11: main (0x55fde9970a61)
  12: __libc_start_main (0x7f6283dcf222)
  13: _start (0x55fde99601ad)
  14: <unknown> (0x0)
~~~
I'm using Arch Linux and I used latest version of `avr-rust` from `avr-support` branch(cea79f77f5eac10eddf63e1e8728fd0e34c5998f).
~~~
➤ rustup -V
rustup 1.16.0
~~~
~~~
➤ xargo -V
xargo 0.3.13
cargo 1.34.0-nightly (865cb7010 2019-02-10)
~~~

1inguini, 2019-02-22:
I added `XARGO_RUST=$HOME/Desktop/rust/rust-avr/src/`, the error message changed into something similar to https://github.com/avr-rust/blink/issues/10
~~~
$ XARGO_RUST_SRC=$HOME/Desktop/rust/rust-avr/src/ rustup run avr-toolchain xargo build --target avr-atmega328p
warning: Patch `rustc-std-workspace-core v1.0.0 (/home/linguini/Desktop/rust/rust-avr/src/tools/rustc-std-workspace-core)` was not used in the crate graph.
Check that the patched package version and available features are compatible
with the dependency requirements. If the patch has a different version from
what is locked in the Cargo.lock file, run `cargo update` to use the new
version. This may also occur with an optional dependency that is not enabled.
   Compiling core v0.0.0 (/home/linguini/Desktop/rust/rust-avr/src/libcore)
Expected either Y or Z register
UNREACHABLE executed at /home/linguini/Desktop/rust/rust-avr/src/llvm/lib/Target/AVR/MCTargetDesc/AVRMCCodeEmitter.cpp:146!
error: Could not compile `core`.

Caused by:
  process didn't exit successfully: `rustc --crate-name core /home/linguini/Desktop/rust/rust-avr/src/libcore/lib.rs --color always --crate-type lib --emit=dep-info,link -C opt-level=3 -C metadata=b87379be5561226e -C extra-filename=-b87379be5561226e --out-dir /tmp/xargo.6HHE9DxXxW1t/target/avr-atmega328p/release/deps --target avr-atmega328p -L dependency=/tmp/xargo.6HHE9DxXxW1t/target/avr-atmega328p/release/deps -L dependency=/tmp/xargo.6HHE9DxXxW1t/target/release/deps --sysroot /home/linguini/.xargo -Z force-unstable-if-unmarked` (signal: 6, SIGABRT: process abort signal)
error: `"cargo" "build" "--release" "--manifest-path" "/tmp/xargo.6HHE9DxXxW1t/Cargo.toml" "--target" "avr-atmega328p" "-p" "core"` failed with exit code: Some(101)
stack backtrace:
   0: error_chain::make_backtrace::h90af62d5c4a6fbd6 (0x5605b78ce16b)
   1: <error_chain::State as core::default::Default>::default::h23f817a122243ac9 (0x5605b78ce235)
   2: <std::process::Command as xargo::extensions::CommandExt>::run::h01346a6b4f35f543 (0x5605b7881d4b)
   3: xargo::sysroot::build::h240260972f2e09ed (0x5605b7889541)
   4: xargo::sysroot::update::h94fd82f52808b493 (0x5605b788d65d)
   5: xargo::main::ha282b52dabbcaa07 (0x5605b787ac7a)
   6: std::rt::lang_start::{{closure}}::h6c099b348d99c7de (0x5605b787e8f2)
   7: std::rt::lang_start_internal::{{closure}}::h53f6e74893a27d59 (0x5605b78f9162)
             at src/libstd/rt.rs:59
      std::panicking::try::do_call::h8945e3e0fe63365a
             at src/libstd/panicking.rs:307
   8: __rust_maybe_catch_panic (0x5605b79061a9)
             at src/libpanic_unwind/lib.rs:102
   9: std::panicking::try::hee49dda4da841cb1 (0x5605b78f9058)
             at src/libstd/panicking.rs:286
  10: std::panic::catch_unwind::h89d0b17e1bb724c9 (0x5605b78ec665)
             at src/libstd/panic.rs:398
      std::rt::lang_start_internal::hcbec2a0475a1169a
             at src/libstd/rt.rs:58
  11: main (0x5605b787ba61)
  12: __libc_start_main (0x7f23af52c222)
  13: _start (0x5605b786b1ad)
  14: <unknown> (0x0)
~~~

nevzheng, 2019-03-06:
I believe that I am encountering the same/similar issue:

Environment:
```
```

Reproduction:

Result:
```
 nevin@manjaro  ~/projects/avr-rust/blink   master  export XARGO_RUST_SRC=/home/nevin/projects/avr-rust/rust/src                                            ✔  947  18:37:25
 nevin@manjaro  ~/projects/avr-rust/blink   master  rustup run avr-toolchain xargo build --target avr-atmega328p --release                                  ✔  961  18:39:16
warning: Patch `rustc-std-workspace-core v1.0.0 (/home/nevin/projects/avr-rust/rust/src/tools/rustc-std-workspace-core)` was not used in the crate graph.
Check that the patched package version and available features are compatible
with the dependency requirements. If the patch has a different version from
what is locked in the Cargo.lock file, run `cargo update` to use the new
version. This may also occur with an optional dependency that is not enabled.
   Compiling core v0.0.0 (/home/nevin/projects/avr-rust/rust/src/libcore)
Expected either Y or Z register
UNREACHABLE executed at /home/nevin/projects/avr-rust/rust/src/llvm/lib/Target/AVR/MCTargetDesc/AVRMCCodeEmitter.cpp:146!
error: Could not compile `core`.

Caused by:
  process didn't exit successfully: `rustc --crate-name core /home/nevin/projects/avr-rust/rust/src/libcore/lib.rs --color always --crate-type lib --emit=dep-info,link -C opt-level=3 -C metadata=9f7c9bd4c8de804c -C extra-filename=-9f7c9bd4c8de804c --out-dir /tmp/xargo.1k4g57QclUqY/target/avr-atmega328p/release/deps --target avr-atmega328p -L dependency=/tmp/xargo.1k4g57QclUqY/target/avr-atmega328p/release/deps -L dependency=/tmp/xargo.1k4g57QclUqY/target/release/deps --sysroot /home/nevin/.xargo -Z force-unstable-if-unmarked` (signal: 6, SIGABRT: process abort signal)
error: `"cargo" "build" "--release" "--manifest-path" "/tmp/xargo.1k4g57QclUqY/Cargo.toml" "--target" "avr-atmega328p" "-p" "core"` failed with exit code: Some(101)
stack backtrace:
   0: error_chain::make_backtrace::h5032a63a5ed44fdd (0x55aef7035ed7)
   1: <error_chain::State as core::default::Default>::default::hc4c04c3c9a6140ef (0x55aef7035f65)
   2: <std::process::Command as xargo::extensions::CommandExt>::run::h22c768f80cad706c (0x55aef7002f60)
   3: xargo::sysroot::build::h90ce42cc6103b881 (0x55aef6ffac94)
   4: xargo::sysroot::update::h94266ac5be482c89 (0x55aef6ffe356)
   5: xargo::main::hffc18a1591dcef6e (0x55aef6fec10b)
   6: std::rt::lang_start::{{closure}}::h21174c8ffdd477e2 (0x55aef6fefa22)
   7: {{closure}} (0x55aef7053712)
             at src/libstd/rt.rs:49
      do_call<closure,i32>
             at src/libstd/panicking.rs:297
   8: __rust_maybe_catch_panic (0x55aef705bd59)
             at src/libpanic_unwind/lib.rs:87
   9: try<i32,closure> (0x55aef705421c)
             at src/libstd/panicking.rs:276
      catch_unwind<closure,i32>
             at src/libstd/panic.rs:388
      lang_start_internal
             at src/libstd/rt.rs:48
  10: main (0x55aef6fecb81)
  11: __libc_start_main (0x7fe5ef43a222)
  12: _start (0x55aef6fe119d)
  13: <unknown> (0x0)
```

dylanmckay, 2019-04-01:
This sounds like avr-rust/rust#/128

Thanks for the bug reports @1inguini and @nevzheng!

stappersg, 2021-01-16:
For which reason is the issue from before the merge of AVRust into upstream Rust as of July 2020?

stappersg, 2021-02-04:
Closing for two reasons
* The long silence
*  AVRust being merged with upstream Rust since July 2020

In case I'm too eager with this close, please do more then clicking on `re-open`, such as creating a fresh issue.

closed 2021-02-04 by stappersg