avr-rust/blink #12: Compilation fails at "rustc --print sysroot"

floworbit, 2019-03-24:
Building and installing rust-avr both completed successfully for me, but when I try to compile the blink example I get the following error:

```
flo@host:~/rust-blink$ rustup run avr-toolchain xargo build --target avr-atmega328p --release
error: `"rustc" "--print" "sysroot"` failed with exit code: Some(1)
stack backtrace:
   0: <no info> (0x55d409d6fb27)
   1: <no info> (0x55d409d6fbb5)
   2: <no info> (0x55d409d2a1ba)
   3: <no info> (0x55d409d2a750)
   4: <no info> (0x55d409d4228e)
   5: <no info> (0x55d409d22412)
   6: <no info> (0x55d409d82ef2)
   7: <no info> (0x55d409da2189)
   8: <no info> (0x55d409d8db5a)
   9: <no info> (0x55d409d43ac4)
  10: <no info> (0x7f126c84109a)
  11: <no info> (0x55d409d171e9)
  12: <no info> (0x0)
```
and when I run "rustc --print sysroot" directly it says
```
flo@host:~/fabac/ex09/rust-src$ rustc --print sysroot
error: couldn't load codegen backend "/home/flo/avr-rust/build/build/x86_64-unknown-linux-gnu/stage1/lib/rustlib/x86_64-unknown-linux-gnu/codegen-backends/librustc_codegen_llvm-llvm.so": "/home/flo/avr-rust/build/build/x86_64-unknown-linux-gnu/stage1/lib/rustlib/x86_64-unknown-linux-gnu/codegen-backends/librustc_codegen_llvm-llvm.so: invalid ELF header"
```

dylanmckay, 2019-04-01:
Would you be able to compile `rustc` with debug mode and copy the stacktrace?

This is an interesting one, it vaguely seems like it could be related to your cross-compiler setup, `rustc --print sysroot` shouldn't even hit the LLVM AVR backend.

floworbit, 2019-04-05:
Sorry I can't do that, now that I have cleaned up everything and recompiled rustc in debug mode everything is working fine.

closed 2019-04-05 by floworbit