avr-rust/blink #13: compiler_builtins fails to compile due to using recently stabilized cfg_target_vendor

mister-walter, 2019-05-13:
I'm not sure if this belongs here or on the main `avr-rust/rust` repo, but just wanted to post this here in case it's helpful to anyone else.

I'm using Xargo on Arch Linux with the latest commit of the `avr-rust/rust` master branch.

I managed to get past avr-rust/rust#128 using hercek's patches, but building `compiler_builtins` failed because of a use of the `cfg_config_vendor` feature. That feature was stabilized earlier this year, but the it looks like the AVR patched Rust version hasn't pulled upstream since the stabilization.

To fix this, I simply modified the downloaded source for `lib.rs` in `compiler_builtins` in the Cargo registry folder (`~/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.12/src/lib.rs` on my machine), adding the missing `#![feature(cfg_target_vendor)]` line to the existing feature flags in that file.

I also get a message that indicates that it might be possible to set a version for `compiler_builtins`, though I wasn't able to figure out how:
```warning: dependency (compiler_builtins) specified without providing a local path, Git repository, or version to use. This will be considered an error in future versions```

The last version of `compiler_builtins` that has the `cfg_target_vendor` feature flag appears to be 0.1.5. ([relevant commit](https://github.com/rust-lang-nursery/compiler-builtins/commit/b570ccfd2d700ba61c8347423fa3c99c01a332e6))

Rahix, 2019-05-26:
I was able to get it working by creating a `Xargo.toml` with the following contents:
```rust
[dependencies.core]
stage = 0

[dependencies.compiler_builtins]
stage = 1
version = "= 0.1.3"
```

dylanmckay, 2019-05-27:
I guess this would be fixed by https://github.com/avr-rust/rust/pull/137

Rahix, 2019-05-28:
Yes, I believe so.  Any rough estimate, when that update is going to hit? / Is there a list of blocking issues that still need work?

dylanmckay, 2019-06-06:
At the moment, the only issue blocking the merge is this bug

> LLVM ERROR: Cannot select: t2: i16 = addrspacecast[1 -> 0] undef:i16

mentioned in [this](https://github.com/avr-rust/rust/pull/137#issuecomment-497946669) comment.

stappersg, 2021-02-04:
> At the moment, the only issue blocking the merge is this bug
> 
> > LLVM ERROR: Cannot select: t2: i16 = addrspacecast[1 -> 0] undef:i16
> 
> mentioned in [this](https://github.com/avr-rust/rust/pull/137#issuecomment-497946669) comment.

That merge did happen.
Closing this issue about  `xargo`.  Lots of things changed since the opening of this issue.

closed 2021-02-04 by stappersg