avr-rust/blink #14: failed to read libcompiler_builtins/Cargo.toml

temoto, 2019-07-29:
OS: Darwin 16.7.0 (MacOS Sierra 10.12.6)

Followed special instructions for mac in avr-rust/rust/readme up to `rustup toolchain link avr-toolchain` **without** `rustup default avr-toolchain`

```
% env |fgrep RUST
RUST_TARGET_PATH=/Users/temoto/dev/github.com/avr-rust/blink
XARGO_RUST_SRC=/Users/temoto/dev/github.com/avr-rust/rust/src

% RUST_BACKTRACE=1 rustup run avr-toolchain xargo build --target avr-atmega328p --release
   Compiling core v0.0.0 (/Users/temoto/dev/github.com/avr-rust/rust/src/libcore)
    Finished release [optimized] target(s) in 43.86s
error: failed to load source for a dependency on `compiler_builtins`

Caused by:
  Unable to update /Users/temoto/dev/github.com/avr-rust/rust/src/libcompiler_builtins

Caused by:
  failed to read `/Users/temoto/dev/github.com/avr-rust/rust/src/libcompiler_builtins/Cargo.toml`

Caused by:
  No such file or directory (os error 2)
error: `"cargo" "build" "--release" "--manifest-path" "/var/folders/5s/j0nv22m55090n3x2h_pc17s80000gn/T/xargo.1ZmNvJ8QDbMP/Cargo.toml" "--target" "avr-atmega328p" "-p" "compiler_builtins"` failed with exit code: Some(101)
stack backtrace:
   0:        0x10e099dae - backtrace::backtrace::trace::h2b288dae4e90468b
   1:        0x10e09a94c - <backtrace::capture::Backtrace as core::default::Default>::default::h550523ee2bda132d
   2:        0x10e09a9cd - backtrace::capture::Backtrace::new::h3c27268eb4bacd81
   3:        0x10e0995f2 - error_chain::make_backtrace::hbe0ec31a030f79de
   4:        0x10e099688 - <error_chain::State as core::default::Default>::default::hf698b87d27c816d2
   5:        0x10e052853 - <std::process::Command as xargo::extensions::CommandExt>::run::hf63ef3ddf82c049c
   6:        0x10e05a29e - xargo::sysroot::build::hf6c5e0460be6e1f5
   7:        0x10e05e408 - xargo::sysroot::update::h62344c6baa369a4f
   8:        0x10e04c299 - xargo::main::h98ed4b12355b82e8
   9:        0x10e067b35 - std::rt::lang_start::{{closure}}::h11c6703c26ec252f
  10:        0x10e0aeb47 - std::panicking::try::do_call::h7aed6cf30733d762
  11:        0x10e0c298e - ___rust_maybe_catch_panic
  12:        0x10e0af41a - std::rt::lang_start_internal::hf650867aab82d8a2
  13:        0x10e04cd0b - _main
RUST_BACKTRACE=1 rustup run avr-toolchain xargo build --target avr-atmega328p  55.70s user 1.97s system 129% cpu 44.621 total
```

ratkins, 2019-10-21:
Did you ever figure this out?

Had you run the `.../rust/configure [...]` and `make && make install` lines before this, and did they succeed? It says it can't find `Cargo.toml` which I understand is generated by `configure`.

temoto, 2019-10-21:
Sorry, I had no further progress with it. rust/configure+make was run and succeeded, yes.

Rahix, 2019-10-21:
`compiler_builtins` is no longer part of the rust sources but instead was factored out into a [separate crate](https://github.com/rust-lang/compiler-builtins).  What version of `xargo` do you have?

stappersg, 2021-02-04:
15 months of silence allowed me to close this issue.

closed 2021-02-04 by stappersg