avr-rust/blink #15: Cannot compile libcore

GreatGodOfFire, 2020-03-23:
I followed instructions for `avr-rust` and `blink` compilation. But I am getting this error:

```
$ RUST_BACKTRACE=1 rustup run avr-toolchain xargo build --target avr-atmega328p --release --verbose
+ "rustc" "rustc" "--print" "sysroot"
+ "rustc" "rustc" "--print" "target-list"
+ RUSTFLAGS="--sysroot /home/greatgodoffire/.xargo -Z force-unstable-if-unmarked"
+ "cargo" "cargo" "build" "--release" "--manifest-path" "/tmp/xargo.MNxWNBlqynYS/Cargo.toml" "--target" "avr-atmega328p" "-v" "-p" "core"
warning: Patch `rustc-std-workspace-alloc v1.0.0 (/home/greatgodoffire/rust/rust/src/tools/rustc-std-workspace-alloc)` was not used in the crate graph.
Patch `rustc-std-workspace-core v1.0.0 (/home/greatgodoffire/rust/rust/src/tools/rustc-std-workspace-core)` was not used in the crate graph.
Check that the patched package version and available features are compatible
with the dependency requirements. If the patch has a different version from
what is locked in the Cargo.lock file, run `cargo update` to use the new
version. This may also occur with an optional dependency that is not enabled.
   Compiling core v0.0.0 (/home/greatgodoffire/rust/rust/src/libcore)
     Running `rustc --crate-name core --edition=2018 /home/greatgodoffire/rust/rust/src/libcore/lib.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C metadata=e4bf3f58269cc549 -C extra-filename=-e4bf3f58269cc549 --out-dir /tmp/xargo.MNxWNBlqynYS/target/avr-atmega328p/release/deps --target avr-atmega328p -L dependency=/tmp/xargo.MNxWNBlqynYS/target/avr-atmega328p/release/deps -L dependency=/tmp/xargo.MNxWNBlqynYS/target/release/deps --sysroot /home/greatgodoffire/.xargo -Z force-unstable-if-unmarked`
error: Unrecognized option: 'json'

error: could not compile `core`.

Caused by:
  process didn't exit successfully: `rustc --crate-name core --edition=2018 /home/greatgodoffire/rust/rust/src/libcore/lib.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C metadata=e4bf3f58269cc549 -C extra-filename=-e4bf3f58269cc549 --out-dir /tmp/xargo.MNxWNBlqynYS/target/avr-atmega328p/release/deps --target avr-atmega328p -L dependency=/tmp/xargo.MNxWNBlqynYS/target/avr-atmega328p/release/deps -L dependency=/tmp/xargo.MNxWNBlqynYS/target/release/deps --sysroot /home/greatgodoffire/.xargo -Z force-unstable-if-unmarked` (exit code: 1)
error: `"cargo" "cargo" "build" "--release" "--manifest-path" "/tmp/xargo.MNxWNBlqynYS/Cargo.toml" "--target" "avr-atmega328p" "-v" "-p" "core"` failed with exit code: Some(101)
stack backtrace:
   0: error_chain::make_backtrace
   1: <error_chain::State as core::default::Default>::default
   2: <std::process::Command as xargo::extensions::CommandExt>::run
   3: xargo::sysroot::build
   4: xargo::sysroot::update
   5: xargo::main_inner
   6: std::rt::lang_start::{{closure}}
   7: std::rt::lang_start_internal::{{closure}}
             at src/libstd/rt.rs:52
      std::panicking::try::do_call
             at src/libstd/panicking.rs:292
   8: __rust_maybe_catch_panic
             at src/libpanic_unwind/lib.rs:78
   9: std::panicking::try
             at src/libstd/panicking.rs:270
      std::panic::catch_unwind
             at src/libstd/panic.rs:394
      std::rt::lang_start_internal
             at src/libstd/rt.rs:51
  10: main
  11: __libc_start_main
  12: _start

```

I'm using Ubuntu 16.04 and I've compiled `avr-rust` yesterday.

```
$ xargo --version
xargo 0.3.19
cargo 1.42.0-nightly (9d32b7b01 2020-01-26)
```

Rahix, 2020-03-30:
I've seen this error a few times before but sadly I don't quite remember what caused it.  I think it was an issue with a `cargo`/`xargo` version too new for the `rustc` version.

What does

```shell-session
rustup run avr-toolchain rustc --version
rustup run avr-toolchain cargo --version
rustup run avr-toolchain xargo --version
```

tell you?

dylanmckay, 2020-03-31:
I've had problems with Xargo recently, can recommend `cargo-xbuild`, a more maintained fork.

Most of these weird toolchain errors that always crop up seem unrelated to AVR specifically, it's a bit of a pain.

GreatGodOfFire, 2020-03-31:
When I use `cargo-xbuild`, I get this error: 
```
$ RUST_BACKTRACE=1 rustup run avr-toolchain cargo xbuild --target avr-atmega328p --release --verbose
build --target avr-atmega328p --release --verbose
\+ "rustc" "rustc" "--print" "sysroot"
\+ "rustc" "rustc" "--print" "target-list"
\+ "/home/greatgodoffire/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/cargo" "/home/greatgodoffire/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/cargo" "rustc" "-p" "alloc" "--release" "--manifest-path" "/tmp/xargo.ypG8Odg7CdkR/Cargo.toml" "--target" "avr-atmega328p" "-v" "--" "-Z" "force-unstable-if-unmarked"
    Updating crates.io index
   Compiling compiler_builtins v0.1.26
   Compiling core v0.0.0 (/home/greatgodoffire/rust/rust/src/libcore)
     Running \`rustc --crate-name build_script_build /home/greatgodoffire/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.26/build.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C opt-level=3 --cfg 'feature="compiler-builtins"' --cfg 'feature="core"' --cfg 'feature="default"' --cfg 'feature="mem"' -C metadata=daaa692540093c0b -C extra-filename=-daaa692540093c0b --out-dir /tmp/xargo.ypG8Odg7CdkR/target/release/build/compiler_builtins-daaa692540093c0b -L dependency=/tmp/xargo.ypG8Odg7CdkR/target/release/deps --cap-lints allow`
     Running `rustc --crate-name core --edition=2018 /home/greatgodoffire/rust/rust/src/libcore/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C metadata=571e7b06dbb5cde9 -C extra-filename=-571e7b06dbb5cde9 --out-dir /tmp/xargo.ypG8Odg7CdkR/target/avr-atmega328p/release/deps --target avr-atmega328p -L dependency=/tmp/xargo.ypG8Odg7CdkR/target/avr-atmega328p/release/deps -L dependency=/tmp/xargo.ypG8Odg7CdkR/target/release/deps\`
error: Unrecognized option: 'json'

error: could not compile \`compiler_builtins\`.

Caused by:
  process didn't exit successfully: \`rustc --crate-name build_script_build /home/greatgodoffire/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.26/build.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C opt-level=3 --cfg 'feature="compiler-builtins"' --cfg 'feature="core"' --cfg 'feature="default"' --cfg 'feature="mem"' -C metadata=daaa692540093c0b -C extra-filename=-daaa692540093c0b --out-dir /tmp/xargo.ypG8Odg7CdkR/target/release/build/compiler_builtins-daaa692540093c0b -L dependency=/tmp/xargo.ypG8Odg7CdkR/target/release/deps --cap-lints allow\` (exit code: 1)
warning: build failed, waiting for other jobs to finish...
error: Unrecognized option: 'json'

error: could not compile \`core\`.

Caused by:
  process didn't exit successfully: \`rustc --crate-name core --edition=2018 /home/greatgodoffire/rust/rust/src/libcore/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C metadata=571e7b06dbb5cde9 -C extra-filename=-571e7b06dbb5cde9 --out-dir /tmp/xargo.ypG8Odg7CdkR/target/avr-atmega328p/release/deps --target avr-atmega328p -L dependency=/tmp/xargo.ypG8Odg7CdkR/target/avr-atmega328p/release/deps -L dependency=/tmp/xargo.ypG8Odg7CdkR/target/release/deps\` (exit code: 1)
error: \`"/home/greatgodoffire/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/cargo" "/home/greatgodoffire/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/cargo" "rustc" "-p" "alloc" "--release" "--manifest-path" "/tmp/xargo.ypG8Odg7CdkR/Cargo.toml" "--target" "avr-atmega328p" "-v" "--" "-Z" "force-unstable-if-unmarked"\` failed with exit code: Some(101)
```

For @Rahix :
$ rustup run avr-toolchain rustc --version
rustc 1.37.0-dev
$rustup run avr-toolchain cargo --version
cargo 1.42.0-nightly (9d32b7b01 2020-01-26)
$ rustup run avr-toolchain xargo --version
xargo 0.3.19
cargo 1.42.0-nightly (9d32b7b01 2020-01-26)

Rahix, 2020-04-06:
@GreatGodOfFire:  For me:

```console-session
bash-5.0$ rustc --version
rustc 1.37.0-dev
bash-5.0$ cargo --version
cargo 1.37.0
bash-5.0$ xargo --version
xargo 0.3.17 (fd55d412234a 2019-11-08)
cargo 1.37.0
```

Your `cargo` is probably too new and I remember needing to explicitly install the above version of xargo for things to work ...

Rahix, 2020-04-06:
We really need avr-rust/rust#162 ...

GreatGodOfFire, 2020-04-18:
I changed the version of Xargo, but I don't know how to change the Cargo version for the avr-toolchain. I changed the standard version of cargo, but for the avr-toolchain, it's still 1.42.0

Smarco10, 2020-05-22:
I @GreatGodOfFire ,

I successfully solved the problem by installing cargo 1.38.0.
To do so, I uninstalled rust:
`rustup self uninstall`
then I installed rust 1.38:
`rustup toolchain install 1.38.0` (you can check cargo version which should be 1.38: `cargo --version`)
then installed xargo:
`cargo install xargo`
I relinked the already compiled toolchain:
`rustup toolchain link avr-toolchain /opt/avr-rust`
In my case I had to manually add cargo to the toolchain avr-toolchain by copying them from toolchain:
`cp $HOME/.rustup/toolchains/1.38.0-x86_64-unknown-linux-gnu/bin/cargo* $HOME/.rustup/toolchains/avr-toolchain/bin/`
Then I successfully build the target by running:
`XARGO_RUST_SRC=$HOME/devel/avr-rust/src/ rustup run avr-toolchain xargo build --target avr-atmega328p --release`

Hoping it will help

HollayHorvath, 2020-06-28:
What worked for me is to install the `cargo` and `xargo` released for the corresponding version of `rustc`. Since my `rustc` version for `avr` was 1.37, I did the following:
`cargo install --version 0.38.0 cargo --force`
`cargo install --version 0.3.16 xargo --force`

I got the versions from comparing the release date of Rust 1.37 and selecting the corresponding versions of `cargo` and `xargo` from crates.io.

I also had to manually modify some code (it was failing on `#![feature(llvm_asm)]` attributes, I changed those to `asm` along with the `llvm_asm!` macros, 3-4 occurrences all together), but after this I have the `elf` binary.

Please keep in mind that this is a horrible hack.

shepmaster, 2020-06-29:
I wouldn't recommend trying our current public branches at all. AVR support is making its way into Rust proper. The `avr-unknown-unknown` target is available in Rust nightly already. It doesn't compile most anything, due to a number of LLVM and Rust changes needed. I'd follow https://github.com/rust-lang/rust/issues/44052 for the latest updates.

GreatGodOfFire, 2020-06-30:
Yeah, I know. It'll be available soon so I'll close this Issue. Thanks for the help.

closed 2020-06-30 by GreatGodOfFire