avr-rust/blink #21: failes to upload on real hardware

djdisodo, 2020-07-30:
`C:\Users\djdis\Documents\blink>avrdude -patmega328p -carduino -P \\.\COM14 -D -b 115200 -Uflash:w:target\avr-atmega328p\release\blink.elf:i -v -v -v

avrdude: Version 5.10, compiled on Jan 19 2010 at 10:45:23
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\WinAVR-20100110\bin\avrdude.conf"

         Using Port                    : \\.\COM14
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATMEGA328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65     5     4    0 no       1024    4      0  3600  3600 0xff 0xff
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 4.4
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e950f
avrdude: safemode read 1, lfuse value: 0
avrdude: safemode read 2, lfuse value: 0
avrdude: safemode read 3, lfuse value: 0
avrdude: safemode: lfuse reads as 0
avrdude: safemode read 1, hfuse value: 0
avrdude: safemode read 2, hfuse value: 0
avrdude: safemode read 3, hfuse value: 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode read 1, efuse value: 0
avrdude: safemode read 2, efuse value: 0
avrdude: safemode read 3, efuse value: 0
avrdude: safemode: efuse reads as 0
avrdude: reading input file "target\avr-atmega328p\release\blink.elf"
avrdude: WARNING: no end of file record found for Intel Hex file "target\avr-atmega328p\release\blink.elf"
avrdude: writing flash (0 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 0 bytes of flash written
avrdude: verifying flash memory against target\avr-atmega328p\release\blink.elf:
avrdude: load data flash data from input file target\avr-atmega328p\release\blink.elf:
avrdude: WARNING: no end of file record found for Intel Hex file "target\avr-atmega328p\release\blink.elf"
avrdude: input file target\avr-atmega328p\release\blink.elf contains 0 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 11.02s

avrdude: verifying ...
avrdude: 0 bytes of flash verified

avrdude: safemode read 1, lfuse value: 0
avrdude: safemode read 2, lfuse value: 0
avrdude: safemode read 3, lfuse value: 0
avrdude: safemode: lfuse reads as 0
avrdude: safemode read 1, hfuse value: 0
avrdude: safemode read 2, hfuse value: 0
avrdude: safemode read 3, hfuse value: 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode read 1, efuse value: 0
avrdude: safemode read 2, efuse value: 0
avrdude: safemode read 3, efuse value: 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK

avrdude done.  Thank you.`

i think this is problem
`avrdude: WARNING: no end of file record found for Intel Hex file "target\avr-atmega328p\release\blink.elf"
avrdude: writing flash (0 bytes):
`

shepmaster, 2020-07-30:
Where did you get that avrdude command line from? [the guide](https://book.avr-rust.com/004-flashing-a-crate-to-chip.html) specifically has you use `:e`, not `:i`

djdisodo, 2020-07-31:
@shepmaster 
:i is used when uploading precompiled arduino sketch
i tried with :e, doesn't works
`invalid file format 'e' in update specifier`

djdisodo, 2020-07-31:
i tried it with config and avrdude from arduino then worked
problem was using avrdude from WinAVR
add
`C:\Program Files (x86)\Arduino\hardware\tools\avr\bin`
to Path
and **don't use WinAVR**

closed 2020-08-01 by djdisodo