avr-rust/blink #24: Failing to build in debug mode

vringar, 2020-08-18:
When running `cargo build -Z build-std=core --target avr-atmega328p.json --release` the code compiles as expected, but when I try to run `cargo build -Z build-std=core --target avr-atmega328p.json` it errors out when trying to link as it fails to find `abort`.

<details>
  <summary> <code>cargo build -Z build-std=core --target avr-atmega328p.json --verbose </code> full output</summary>
  
```
cargo build -Z build-std=core --target avr-atmega328p.json --verbose                                                                                                                                                      master [7c58f52]
       Fresh bitflags v0.7.0
       Fresh lazy_static v0.2.11
       Fresh core v0.0.0 (/home/stefan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
       Fresh xml-rs v0.4.1
       Fresh rustc-std-workspace-core v1.99.0 (/home/stefan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
       Fresh xmltree v0.4.0
       Fresh compiler_builtins v0.1.32
       Fresh avr-mcu v0.2.6
       Fresh ruduino v0.2.0
   Compiling blink v0.1.0 (/home/stefan/projects/rust/blink)
     Running `rustc --crate-name blink --edition=2018 src/main.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -Cembed-bitcode=no -C debuginfo=2 -C metadata=dbd5bc0bab399880 -C extra-filename=-dbd5bc0bab399880 --out-dir /home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps --target /home/stefan/projects/rust/blink/avr-atmega328p.json -C incremental=/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/incremental -L dependency=/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps -L dependency=/home/stefan/projects/rust/blink/target/debug/deps --extern 'noprelude:compiler_builtins=/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/libcompiler_builtins-1047601e27e05202.rlib' --extern 'noprelude:core=/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/libcore-ef7fa51ce8a961de.rlib' --extern ruduino=/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/libruduino-99c4dc9b63733f14.rlib -Z unstable-options`
error: linking with `avr-gcc` failed: exit code: 1
  |
  = note: "avr-gcc" "-mmcu=atmega328p" "-L" "/home/stefan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.1ca5aaqj6bdnuex4.rcgu.o" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.1ebk1l9v3hjau2n1.rcgu.o" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.1fndbknct4lu4wlz.rcgu.o" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.2cv8u6cottlyccro.rcgu.o" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.2zf72k7r92954j3z.rcgu.o" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.3it9xiduxr7ti8mx.rcgu.o" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.3kjq30baaid37pjd.rcgu.o" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.4gq8agxkxhg3wz0k.rcgu.o" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.4hgcoc9n8q9um29z.rcgu.o" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.56sqtto8s98qrgy4.rcgu.o" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.5uaamws58ocokst.rcgu.o" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.quw4540n0ektdc4.rcgu.o" "-o" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.elf" "-Wl,--gc-sections" "-no-pie" "-nodefaultlibs" "-L" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps" "-L" "/home/stefan/projects/rust/blink/target/debug/deps" "-L" "/home/stefan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "-Wl,--start-group" "-Wl,-Bstatic" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/libruduino-99c4dc9b63733f14.rlib" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/librustc_std_workspace_core-c75f23490d0e2274.rlib" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/libcore-ef7fa51ce8a961de.rlib" "-Wl,--end-group" "/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/libcompiler_builtins-1047601e27e05202.rlib" "-Wl,-Bdynamic" "-lgcc"
  = note: /usr/bin/avr-ld: /home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.3it9xiduxr7ti8mx.rcgu.o: in function `core::ptr::swap_nonoverlapping_bytes':
          /home/stefan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ptr/mod.rs:484: undefined reference to `abort'
          /usr/bin/avr-ld: /home/stefan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ptr/mod.rs:506: undefined reference to `abort'
          /usr/bin/avr-ld: /home/stefan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ptr/mod.rs:512: undefined reference to `abort'
          /usr/bin/avr-ld: /home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.5uaamws58ocokst.rcgu.o: in function `core::ptr::swap_nonoverlapping':
          /home/stefan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ptr/mod.rs:437: undefined reference to `abort'
          /usr/bin/avr-ld: /home/stefan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ptr/mod.rs:437: undefined reference to `abort'
          /usr/bin/avr-ld: /home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.5uaamws58ocokst.rcgu.o:/home/stefan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ptr/mod.rs:442: more undefined references to `abort' follow
          /usr/bin/avr-ld: /home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.quw4540n0ektdc4.rcgu.o: in function `core::intrinsics::copy_nonoverlapping':
          /home/stefan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/intrinsics.rs:2150: undefined reference to `memcpy'
          /usr/bin/avr-ld: /home/stefan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/intrinsics.rs:2145: undefined reference to `abort'
          /usr/bin/avr-ld: /home/stefan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/intrinsics.rs:2145: undefined reference to `abort'
          /usr/bin/avr-ld: /home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/blink-dbd5bc0bab399880.quw4540n0ektdc4.rcgu.o: in function `core::intrinsics::is_aligned_and_not_null':
          /home/stefan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/intrinsics.rs:2033: undefined reference to `abort'
          /usr/bin/avr-ld: /home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/libcore-ef7fa51ce8a961de.rlib(core-ef7fa51ce8a961de.core.8suzmktt-cgu.3.rcgu.o): in function `core::intrinsics::is_aligned_and_not_null':
          /home/stefan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/str/lossy.rs:104: undefined reference to `abort'
          /usr/bin/avr-ld: /home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/libcore-ef7fa51ce8a961de.rlib(core-ef7fa51ce8a961de.core.8suzmktt-cgu.3.rcgu.o): in function `core::intrinsics::is_nonoverlapping':
          /home/stefan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/str/lossy.rs:104: undefined reference to `abort'
          /usr/bin/avr-ld: /home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/libcore-ef7fa51ce8a961de.rlib(core-ef7fa51ce8a961de.core.8suzmktt-cgu.3.rcgu.o):/home/stefan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/str/lossy.rs:106: more undefined references to `abort' follow
[... cut ...] pendency=/home/stefan/projects/rust/blink/target/debug/deps --extern 'noprelude:compiler_builtins=/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/libcompiler_builtins-1047601e27e05202.rlib' --extern 'noprelude:core=/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/libcore-ef7fa51ce8a961de.rlib' --extern ruduino=/home/stefan/projects/rust/blink/target/avr-atmega328p/debug/deps/libruduino-99c4dc9b63733f14.rlib -Z unstable-options` (exit code: 1)

```
  
</details>

Rahix, 2020-08-18:
This is most likely because the debug binary is too big and the linker can't fit all of it in flash.  Either just always build in release or tweak the debug profile to use some optimization as well.  E.g. I'm using this:

```toml
[profile.dev]
panic = "abort"
lto = true
opt-level = "s"

[profile.release]
panic = "abort"
codegen-units = 1
debug = true
lto = true
opt-level = "s"
```

dylanmckay, 2020-10-29:
Thanks for writing that up @Rahix - indeed, debug codegen size is very large and release mode is basically a requirement.

I'm going to go ahead and close this

closed 2020-10-29 by dylanmckay