avr-rust/blink #25: SIGSEGV when compiling on latest nightly

bacongobbler, 2020-10-09:
This issue has reported upstream at https://github.com/rust-lang/rust/issues/76930. Following the AVR Rust book, the program cannot compile due to a segmentation fault when compiling the `core` crate on the latest nightly build.

```
~/code/blink ><> cargo build -Z build-std=core --target avr-atmega328p.json --release
   Compiling core v0.0.0 (/home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling rustc-std-workspace-core v1.99.0 (/home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling compiler_builtins v0.1.35
   Compiling ruduino v0.2.0
error: could not compile `core`

Caused by:
  process didn't exit successfully: `rustc --crate-name core --edition=2018 /home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no -C metadata=775b9b151bd266b1 -C extra-filename=-775b9b151bd266b1 --out-dir /home/bacongobbler/code/blink/target/avr-atmega328p/release/deps --target /home/bacongobbler/code/blink/avr-atmega328p.json -Z force-unstable-if-unmarked -L dependency=/home/bacongobbler/code/blink/target/avr-atmega328p/release/deps -L dependency=/home/bacongobbler/code/blink/target/release/deps --cap-lints allow` (signal: 11, SIGSEGV: invalid memory reference)
warning: build failed, waiting for other jobs to finish...
error: build failed
```

Using the debug target also fails with more information:

```
~/code/blink ><> cargo build -Z build-std=core --target avr-atmega328p.json
   Compiling blink v0.1.0 (/home/bacongobbler/code/blink)
error: linking with `avr-gcc` failed: exit code: 1
  |
  = note: "avr-gcc" "-mmcu=atmega328p" "-L" "/home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.14gicplk3a8rd2mf.rcgu.o" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.14qeofa20egw9nwf.rcgu.o" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.1wmmysfy11urhzf0.rcgu.o" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.24kbbo2eafqv76ap.rcgu.o" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.2mrol613m7r8fe4h.rcgu.o" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.2ygh13f44y84rejg.rcgu.o" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.3gpysx65vrtx1mw3.rcgu.o" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.3hgcd7rk5gitckx.rcgu.o" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.4fk979zh9rz3tdvf.rcgu.o" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.4xxrwed9dq0c99sc.rcgu.o" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.5c9h7xhuo2dh8daz.rcgu.o" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.fooo9so7gsdl2oq.rcgu.o" "-o" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.elf" "-Wl,--gc-sections" "-nodefaultlibs" "-L" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps" "-L" "/home/bacongobbler/code/blink/target/debug/deps" "-L" "/home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "-Wl,--start-group" "-Wl,-Bstatic" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/libruduino-320958a79d2f2860.rlib" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/librustc_std_workspace_core-29c9d0aa85cbe977.rlib" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/libcore-69e33a9af4453194.rlib" "-Wl,--end-group" "/home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/libcompiler_builtins-f1cb32ff07da7789.rlib" "-Wl,-Bdynamic" "-lgcc"
  = note: /home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.3gpysx65vrtx1mw3.rcgu.o: In function `core::intrinsics::is_nonoverlapping::h95d5cae38d8a279d':
          /home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/intrinsics.rs:1754: undefined reference to `abort'
          /home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/intrinsics.rs:1754: undefined reference to `abort'
          /home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.3gpysx65vrtx1mw3.rcgu.o: In function `core::intrinsics::copy_nonoverlapping::h106c315c8151203b':
          /home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/intrinsics.rs:1860: undefined reference to `memcpy'
          /home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/intrinsics.rs:1855: undefined reference to `abort'
          /home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/intrinsics.rs:1855: undefined reference to `abort'
          /home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.3gpysx65vrtx1mw3.rcgu.o: In function `core::intrinsics::is_aligned_and_not_null::h4f686583b9419b1a':
          /home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/intrinsics.rs:1745: undefined reference to `abort'
          /home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.4fk979zh9rz3tdvf.rcgu.o: In function `core::ptr::swap_nonoverlapping_bytes::h263f7c0912ae4c3d':
          /home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ptr/mod.rs:472: undefined reference to `abort'
          /home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ptr/mod.rs:494: undefined reference to `abort'
          /home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/blink-40c8e0f02000ab5f.4fk979zh9rz3tdvf.rcgu.o:/home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ptr/mod.rs:500: more undefined references to `abort' follow
          /home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/libcore-69e33a9af4453194.rlib(core-69e33a9af4453194.core.a37ui6x9-cgu.13.rcgu.o): In function `core::intrinsics::copy_nonoverlapping::h12295594d5eb04a5':
          /home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/intrinsics.rs:1860: undefined reference to `memcpy'
          /home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/intrinsics.rs:1855: undefined reference to `abort'
          /home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/intrinsics.rs:1855: undefined reference to `abort'
          /home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/libcore-69e33a9af4453194.rlib(core-69e33a9af4453194.core.a37ui6x9-cgu.3.rcgu.o): In function `core::panicking::panic::hf3e1dc38e8693d18':
          /home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/num/bignum.rs:287: undefined reference to `abort'
          /home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/libcore-69e33a9af4453194.rlib(core-69e33a9af4453194.core.a37ui6x9-cgu.3.rcgu.o): In function `core::panicking::panic_fmt::hbbf5d4a80efa2a13':
          /home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/panicking.rs:85: undefined reference to `abort'
          /home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/libcore-69e33a9af4453194.rlib(core-69e33a9af4453194.core.a37ui6x9-cgu.9.rcgu.o): In function `core::option::Option$LT$T$GT$::unwrap::h73d81fcba2eb990c':
[... cut ...]
          /home/bacongobbler/code/blink/target/avr-atmega328p/debug/deps/libcore-69e33a9af4453194.rlib(core-69e33a9af4453194.core.a37ui6x9-cgu.9.rcgu.o):/home/bacongobbler/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/option.rs:(.text._ZN4core6option15Option$LT$T$GT$6unwrap17h73d81fcba2eb990cE+0x5e): more undefined references to `abort' follow
          collect2: error: ld returned 1 exit status
          

error: aborting due to previous error

error: could not compile `blink`

To learn more, run the command again with --verbose.
```

bacongobbler, 2020-10-09:
Looks like things work when I add `-lc` to the `late-link-args` parameter, partially reverting https://github.com/avr-rust/blink/commit/76a28c425df6bd60669fdaa4164267fdabfe2153.

bacongobbler, 2020-10-09:
Hmm... I cannot appear to reproduce this when I attempt to compile the program with the provided Dockerfile. Perhaps there's an issue with the linker being selected locally. I'll do some more research.

bacongobbler, 2020-10-09:
Err, scratch that. It also appears to be an issue with Dockerfile.ci:

```
~/code/blink ><> docker build --file Dockerfile.ci .
...
Successfully built 066d400f55e3
~/code/blink ><> docker run 066d400f55e3
    Updating crates.io index
 Downloading crates ...
  Downloaded lazy_static v0.2.11
  Downloaded ruduino v0.2.0
  Downloaded xml-rs v0.4.1
  Downloaded xmltree v0.4.0
  Downloaded bitflags v0.7.0
  Downloaded avr-mcu v0.2.6
 Downloading crates ...
  Downloaded getopts v0.2.21
  Downloaded addr2line v0.13.0
  Downloaded compiler_builtins v0.1.35
  Downloaded miniz_oxide v0.4.0
  Downloaded adler v0.2.3
  Downloaded cc v1.0.60
  Downloaded hashbrown v0.9.0
  Downloaded unicode-width v0.1.8
  Downloaded cfg-if v0.1.10
  Downloaded libc v0.2.79
  Downloaded rustc-demangle v0.1.16
  Downloaded object v0.20.0
  Downloaded gimli v0.22.0
   Compiling bitflags v0.7.0
   Compiling lazy_static v0.2.11
   Compiling compiler_builtins v0.1.35
   Compiling core v0.0.0 (/home/avr-rust/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling xml-rs v0.4.1
   Compiling xmltree v0.4.0
   Compiling avr-mcu v0.2.6
   Compiling ruduino v0.2.0
   Compiling rustc-std-workspace-core v1.99.0 (/home/avr-rust/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
error: could not compile `core`

Caused by:
  process didn't exit successfully: `rustc --crate-name core --edition=2018 /home/avr-rust/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no -C metadata=5713a875ada81601 -C extra-filename=-5713a875ada81601 --out-dir /code/blink/target/avr-atmega328p/release/deps --target /code/blink/avr-atmega328p.json -Z force-unstable-if-unmarked -L dependency=/code/blink/target/avr-atmega328p/release/deps -L dependency=/code/blink/target/release/deps --cap-lints allow` (signal: 11, SIGSEGV: invalid memory reference)
warning: build failed, waiting for other jobs to finish...
error: build failed
```

So it appears this is not just localized to my system, and it does appear to be related to an issue with the nightlies.

kakkun61, 2020-10-10:
I had a similar error `STATUS_ACCESS_VIOLATION` at `error: could not compile core` with the release flag on Windows.

bergercookie, 2020-10-11:
> Looks like things work when I add -lc to the late-link-args parameter, partially reverting 76a28c4. That only works for the debug target, though. Adding --release still results in a segmentation fault.

Also had the same issue in my setup, this also seems to fix it  :+1:

erdeszt, 2020-10-19:
I had the same issue and got it working with the `-lc` workaround and debug mode. 
By changing the `[profile.dev]` config to:
```
opt-level = "s"
debug = false
debug-assertions = false
overflow-checks = false
lto = false
panic = "abort"
```
I got the output size down from ~700K to 8K which is acceptable to me for now.

ZiCog, 2020-10-30:
I have the same problem. With or without erdezt's changes to [profile.dev]:

```
rustup toolchain install nightly
 2075  rustup component add rust-src --toolchain nightly
 2076  sudo apt-get install binutils gcc-avr avr-libc avrdude
 2077  cd ..
 2078  git clone https://github.com/avr-rust/blink.git
 2079  cd blink/
 2080  rustup override set nightly
 2081  cargo build -Z build-std=core --target avr-atmega328p.json --release
 2082  cd ..
 2083  git clone https://github.com/avr-rust/template-bin.git
 2084  cargo build --target avr-atmega328p.json -Z build-std=core --release
 2085  cd template-bin/
 2086  cargo build --target avr-atmega328p.json -Z build-std=core --release
 2087  history
michael@monster:/mnt/c/Users/michael/template-bin$ cargo build --target avr-atmega328p.json -Z build-std=core --release
   Compiling core v0.0.0 (/home/michael/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling rustc-std-workspace-core v1.99.0 (/home/michael/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling compiler_builtins v0.1.35
error: could not compile `core`

Caused by:
  process didn't exit successfully: `rustc --crate-name core --edition=2018 /home/michael/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no -C metadata=f3eb45e17275f6d1 -C extra-filename=-f3eb45e17275f6d1 --out-dir /mnt/c/Users/michael/template-bin/target/avr-atmega328p/release/deps --target /mnt/c/Users/michael/template-bin/avr-atmega328p.json -Z force-unstable-if-unmarked -L dependency=/mnt/c/Users/michael/template-bin/target/avr-atmega328p/release/deps -L dependency=/mnt/c/Users/michael/template-bin/target/release/deps --cap-lints allow` (signal: 11, SIGSEGV: invalid memory reference)
warning: build failed, waiting for other jobs to finish...
error: build failed
```

nathantypanski, 2020-11-18:
@erdeszt's profile works reliably. Without it, I get the following logs:

```

% cargo build -Z build-std=core --target avr-atmega328p.json                                     Updating crates.io index
   Compiling blink v0.1.0 ($HOME/prj/avr-rust-blink)
    Finished dev [unoptimized + debuginfo] target(s) in 2.74s
nathan@whitebook ~/prj/avr-rust-blink (git)-[master] % cargo build -Z build-std=core --target avr-atmega328p.json
    Updating crates.io index
   Compiling blink v0.1.0 ($HOME/prj/avr-rust-blink)
nathan@whitebook ~/prj/avr-rust-blink (git)-[master] % cargo build -Z build-std=core --target avr-atmega328p.json release
130 nathan@whitebook ~/prj/avr-rust-blink (git)-[master] % cargo build -Z build-std=core --target avr-atmega328p.json --release
    Updating crates.io index
   Compiling core v0.0.0 ($HOME/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling rustc-std-workspace-core v1.99.0 ($HOME/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling compiler_builtins v0.1.36 ($HOME/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/vendor/compiler_builtins)
   Compiling ruduino v0.2.0
error: could not compile `core`

Caused by:
  process didn't exit successfully: `rustc --crate-name core --edition=2018 $HOME/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no -C metadata=2a7c2bfff040f346 -C extra-filename=-2a7c2bfff040f346 --out-dir $HOME/prj/avr-rust-blink/target/avr-atmega328p/release/deps --target $HOME/prj/avr-rust-blink/avr-atmega328p.json -Z force-unstable-if-unmarked -L dependency=$HOME/prj/avr-rust-blink/target/avr-atmega328p/release/deps -L dependency=$HOME/prj/avr-rust-blink/target/release/deps --cap-lints allow` (signal: 11, SIGSEGV: invalid memory reference)
warning: build failed, waiting for other jobs to finish...
error: build failed
cargo build -Z build-std=core --target avr-atmega328p.json --release  16.45s user 0.27s system 147% cpu 11.328 total
101 nathan@whitebook ~/prj/avr-rust-blink (git)-[master] % git stash
Saved working directory and index state WIP on master: 5f541fe Replace link to existing avr-rust/rust repository with a link to AVR book
nathan@whitebook ~/prj/avr-rust-blink (git)-[master] % cargo build -Z build-std=core --target avr-atmega328p.json --release
    Updating crates.io index
   Compiling core v0.0.0 ($HOME/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling rustc-std-workspace-core v1.99.0 ($HOME/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling compiler_builtins v0.1.36 ($HOME/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/vendor/compiler_builtins)
   Compiling ruduino v0.2.0
error: could not compile `core`

Caused by:
  process didn't exit successfully: `rustc --crate-name core --edition=2018 $HOME/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no -C metadata=2a7c2bfff040f346 -C extra-filename=-2a7c2bfff040f346 --out-dir $HOME/prj/avr-rust-blink/target/avr-atmega328p/release/deps --target $HOME/prj/avr-rust-blink/avr-atmega328p.json -Z force-unstable-if-unmarked -L dependency=$HOME/prj/avr-rust-blink/target/avr-atmega328p/release/deps -L dependency=$HOME/prj/avr-rust-blink/target/release/deps --cap-lints allow` (signal: 11, SIGSEGV: invalid memory reference)
warning: build failed, waiting for other jobs to finish...
error: build failed
cargo build -Z build-std=core --target avr-atmega328p.json --release  15.59s user 0.31s system 172% cpu 9.248 total
101 nathan@whitebook ~/prj/avr-rust-blink (git)-[master] % cargo build -Z build-std=core --target avr-atmega328p.json
    Updating crates.io index
   Compiling blink v0.1.0 ($HOME/prj/avr-rust-blink)
error: linking with `avr-gcc` failed: exit code: 1
  |
  = note: "avr-gcc" "-mmcu=atmega328p" "-L" "$HOME/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "$HOME/prj/avr-rust-blink/target/avr-atmega328p/debug/deps/blink-5fec6d403ded3574.1ep73grf7jrprhhi.rcgu.o" "$HOME/prj/avr-rust-blink/target/avr-atmega328p/debug/deps/blink-5fec6d403ded3574.1lhslcthlh80x9xq.rcgu.o" "$HOME/prj/avr-rust-blink/target/avr-atmega328p/debug/deps/blink-5fec6d403ded3574.2kf2mvpvzb8l8vp1.rcgu.o" "$HOME/prj/avr-rust-blink/target/avr-atmega328p/debug/deps/blink-5fec6d403ded3574.2o5dole90y9859fc.rcgu.o" "$HOME/prj/avr-rust-blink/target/avr-atmega328p/debug/deps/blink-5fec6d403ded3574.2rm8ekev70tidj4b.rcgu.o" "$HOME/prj/avr-rust-blink/target/avr-atmega328p/debug/deps/blink-5fec6d403ded3574.38efr3iu7cjqce9v.rcgu.o" "$HOME/prj/avr-rust-blink/target/avr-atmega328p/debug/deps/blink-5fec6d403ded3574.3sqc7xex4l7fg7nm.rcgu.o" "$HOME/prj/avr-rust-blink/target/avr-atmega328p/debug/deps/blink-5fec6d403ded3574.3u71q1fosg4kcdmt.rcgu.o" "$HOME/prj/avr-rust-blink/target/avr-atmega328p/debug/deps/blink-5fec6d403ded3574.3zuqdnkbj3wj399a.rcgu.o" "$HOME/prj/avr-rust-blink/target/avr-atmega328p/debug/deps/blink-5fec6d403ded3574.4arwg6rhk2d46cl6.rcgu.o" "$HOME/prj/avr-rust-bli [... cut ...]
          /usr/bin/avr-ld: $HOME/prj/avr-rust-blink/target/avr-atmega328p/debug/deps/libcore-298146722e7b311b.rlib(core-298146722e7b311b.core.9e76zoye-cgu.15.rcgu.o):$HOME/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/panicking.rs:92: more undefined references to `abort' follow
          collect2: error: ld returned 1 exit status

error: aborting due to previous error

error: could not compile `blink`

To learn more, run the command again with --verbose.
```

timbell87, 2020-11-24:
Slight modification of @erdeszt's workaround because I really just couldn't do without that release build (Cargo.toml in my case):
```
[profile.dev]
opt-level = "s"
debug-assertions = false

[profile.release]
opt-level = "s"
```
I've tested this modification on the dockerfile provided, it works on that and will also build for release.
Output sizes if anyone is interested:
elf: 11.8kb(debug), 6.6kb(release)
binary: 260 bytes (debug and release are the same size)

stappersg, 2022-06-05:
Waits for _seems to be fixed, I can't reprocuce the `SIGSEGV`_ or s simular update.

stappersg, 2022-06-05:
Waits for _seems to be fixed, I can't reprocuce the `SIGSEGV`_ or a simular update.

stappersg, 2022-06-05:
https://github.com/rust-lang/rust/issues/76930 being solved allowed me to close this is issue.
(Feel free to report a fresh _SIGSEGV when compiling on latest nightly_)

closed 2022-06-05 by stappersg