avr-rust/blink #29: LLVM ERROR: Not supported instr: <MCInst 258 <MCOperand Reg:1> <MCOperand Imm:15> <MCOperand Reg:45>>

stappersg, 2021-01-20:
This issue is a pointer to https://github.com/rust-lang/compiler-builtins/issues/400
it is about
```
"LLVM ERROR: Not supported instr" on avr-unknown-gnu-atmega328
```

stappersg, 2021-01-31:
See also https://github.com/Rahix/avr-hal/issues/124#issuecomment-766841616 for how to use _nighty_.

jaredjn, 2021-02-25:
just change to this version (I am on windows)

`rustup default nightly-2021-01-07-x86_64-pc-windows-msvc 
rustup default nightly-2021-01-10-x86_64-pc-windows-msvc
cargo build -Z build-std=core --target avr-atmega328p.json --release`

and you will be back in business.

patrickelectric, 2021-03-15:
I'm having the same problem with `cargo 1.52.0-nightly (32da9eaa5 2021-03-13)`
```
cargo +nightly build -Z build-std=core --target avr-atmega328p.json --release
...
     Running `rustc --crate-name avr_delay --edition=2018 /home/patrick/.cargo/registry/src/github.com-1ecc6299db9ec823/avr_delay-0.3.1/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no -C metadata=1e5ca241312447fe -C extra-filename=-1e5ca241312447fe --out-dir /tmp/blink/target/avr-atmega328p/release/deps --target /tmp/blink/avr-atmega328p.json -L dependency=/tmp/blink/target/avr-atmega328p/release/deps -L dependency=/tmp/blink/target/release/deps --extern avr_config=/tmp/blink/target/avr-atmega328p/release/deps/libavr_config-a37de7453a77e75d.rmeta --extern 'noprelude:compiler_builtins=/tmp/blink/target/avr-atmega328p/release/deps/libcompiler_builtins-442a0f150b34505f.rmeta' --extern 'noprelude:core=/tmp/blink/target/avr-atmega328p/release/deps/libcore-ec1759d2286940d7.rmeta' -Z unstable-options --cap-lints allow`
LLVM ERROR: Not supported instr: <MCInst 296 <MCOperand Reg:1> <MCOperand Imm:15> <MCOperand Reg:43>>
error: could not compile `compiler_builtins`
```

TheNotary, 2021-04-17:
Don't forget to:

```
rustup override unset
```

Since the README has you lock in the latest nightly.  That override was soo confusing, I don't really work with rust but in one of the issues someone mentioned using a rust-toolchain file in the root.  Can we do this instead?

(`rust-toolchain`)
```
[toolchain]
channel = "nightly-2021-01-07"
components = ["rust-src"]
```

stappersg, 2021-04-18:
> Can we do this instead?

That _this_ is proposed in #32 and might be inspired by https://github.com/Rahix/avr-hal/issues/124#issuecomment-795585670

chinoto, 2021-04-19:
@stappersg Nah, I was aware of rust-toolchain beforehand and didn't like having to include `+nightly-2021-01-07` (and the avr hertz and target arch) each time, but I had to lookup how to do the TOML variant so it would automatically fetch the rust-src too. Reminds me of "convergent evolution", we naturally came to the same solutions because they work well.

stappersg, 2022-05-21:
https://github.com/rust-lang/compiler-builtins/issues/400 is reported as being closed.

Next challenge is the `llvm_asm!` to `asm!` switch. Yes that is our #38 ...

stappersg, 2022-06-08:
#38 is fixed, _blink_ does compile again.

closed 2022-06-08 by stappersg