avr-rust/blink #31: similar name exists in the module: `atmega328p`

stappersg, 2021-01-31:
Reporting my expirement
```text
$ cargo +nightly-2021-01-07 build -Z build-std=core --target avr-atmega328p.json --release
   Compiling blink v0.1.0 (/home/gs0604/src/rust/RustAVR/blink)
error[E0432]: unresolved import `ruduino::cores::atmega328`
 --> src/main.rs:6:5
  |
6 | use ruduino::cores::atmega328 as avr_core;
  |     ^^^^^^^^^^^^^^^^---------^^^^^^^^^^^^
  |     |               |
  |     |               help: a similar name exists in the module: `atmega328p`
  |     no `atmega328` in `cores`
```

The `+nightly-2021-01-07` is because #29,  the `README.md` says.
```
cargo build -Z build-std=core --target avr-atmega328p.json --release
```

stappersg, 2021-01-31:
What I did to get a succesfull build:
```diff
--- a/src/main.rs
+++ b/src/main.rs
@@ -3,7 +3,7 @@
 #![no_std]
 #![no_main]
 
-use ruduino::cores::atmega328 as avr_core;
+use ruduino::cores::atmega328p as avr_core;
 use ruduino::Register;
 
 use avr_core::{DDRB, PORTB};
```

I don't know how good  or how evil that is.

Thing I'm trying to say:   `blink` did not compile for me, I had to make a modification to get a `target/avr-atmega328p/release/blink.elf`.

Note: How good that `blink.elf` is, is currently unknown.

dylanmckay, 2021-02-01:
I recently updated the `main.rs` file so that rather than hardcoding the `atmega328`, it uses `use ruduino::cores::current::{DDRB, PORTB};`, using `ruduino` to detect the device that is being targeted. I suspect this issue is fixed now, specifically in cfb84512.

stappersg, 2021-02-14:
Reporting a clean compile, taking #29 workaround in account.
```text
gs0604@trancilo:~/src/rust/RustAVR/blink
$ cargo +nightly-2021-01-07 build -Z build-std=core --target avr-atmega328p.json --release
   Compiling proc-macro2 v1.0.24
   Compiling unicode-xid v0.2.1
   Compiling syn v1.0.60
   Compiling compiler_builtins v0.1.36
   Compiling core v0.0.0 (/home/gs0604/.rustup/toolchains/nightly-2021-01-07-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling json v0.12.4
   Compiling xml-rs v0.8.3
   Compiling lazy_static v1.4.0
   Compiling target-cpu-fetch v0.1.3
   Compiling xmltree v0.10.2
   Compiling target-cpu-macro v0.1.3
   Compiling quote v1.0.8
   Compiling avr-mcu v0.3.5
   Compiling ruduino v0.3.2
   Compiling const_env_impl--value v0.1.2
   Compiling const_env--value v0.1.2
   Compiling rustc-std-workspace-core v1.99.0 (/home/gs0604/.rustup/toolchains/nightly-2021-01-07-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling avr-config v1.0.0
   Compiling avr-std-stub v1.0.2
   Compiling avr-config v2.0.1
   Compiling avr_delay v0.3.1
   Compiling blink v0.1.0 (/home/gs0604/src/rust/RustAVR/blink)
    Finished release [optimized] target(s) in 7.26s
gs0604@trancilo:~/src/rust/RustAVR/blink
$ 
```

closed 2021-02-14 by stappersg