avr-rust/blink #33: Simplify cargo build invocation

chinoto, 2021-04-03:
Maybe it is better that users have to type in the target and build-std flags so they become aware of their necessity, but otherwise they could be omitted with the use of `.cargo/config.toml`. Perhaps a note about them in the README would be enough?

```toml
[build]
target = "avr-atmega328p.json"

[unstable]
build-std = ["core"]

# Cargo versions before 2021-02-23 won't recognize this: https://github.com/rust-lang/cargo/pull/9175
[env]
AVR_CPU_FREQUENCY_HZ = 16000000
```

stappersg, 2022-06-05:
Looks good to me (and I agree on _someone will someday provide a merge request_).

closed 2022-06-06 by stappersg

stappersg, 2022-06-06:
Now we have #41 and would to know on which platform it works.  ( I'm on Debian with `rustup update`d. )

PR #40 (2022-06-06): Add .cargo/config.toml to simplify build command
Resolves #33