avr-rust/blink #34: blinking to fast

vingeni, 2021-04-04:
Hi, the led is blinking too fast.
This is probably due to this [bug on avr-delay](https://github.com/avr-rust/delay/issues/11) ?

It worked well with the previous "hacky loop"

crazcalm, 2021-08-21:
@vingeni I am not sure if you are still using Rust to mess with your Arduino, but there is a [avr-hal](https://github.com/Rahix/avr-hal) crate. Their blinking example works for me.

darionesi, 2021-10-02:
I've added 

```toml
[profiles.release]
lto = true
```

as per [this suggestion](https://www.reddit.com/r/rust/comments/mjkp9v/comment/gx7hrbh/) I've found on reddit and now the code in this repo makes the arduino blink as expected

bmgxyz, 2021-12-12:
That suggestion worked for me, although it looks like the correct key is `profile` (singular), not `profiles` (plural).

I used:

```toml
[profile.release]
lto = true
```

See [this section](https://doc.rust-lang.org/cargo/reference/profiles.html) in The Cargo Book.

PR #39 (2022-06-05): Enable LTO for release builds to fix blink example
The LED blinks much too rapidly without LTO enabled.
With LTO enabled, the LED blinks 1 second on, 1 second off.

Fixes #34.
commit 6b506d21 (2022-06-05): Enable LTO for release builds to fix blink example

The LED blinks much too rapidly without LTO enabled.
With LTO enabled, the LED blinks 1 second on, 1 second off.

Fixes #34.
closed 2022-06-05 by stappersg