avr-rust/blink #36: error: "/usr/lib/rustlib/src/rust/Cargo.lock" does not exist

mjs717, 2021-07-07:
Maybe somebody can help me out with some pointers on what's probably a very basic question/problem. I'm very new to Rust, so my knowledge of the ecosystem is very light, and it's been tough to try to track down what I'm missing. I was hoping to play with Rust in an embedded environment, so that's where I've jumped right into. I'm trying to get the "blink" program up and running on an Arduino Uno that I have laying around. Unfortunately, I can't even get past step 1 (i.e., I can't get it to compile). Again, this is probably something basic I'm missing, so don't assume that it's an intricate issue. :)

I keep getting this error, and nothing I seem to do clears it up:

error: "/usr/lib/rustlib/src/rust/Cargo.lock" does not exist, unable to build with the standard library, try:
        rustup component add rust-src

That path does exist. There's a Cargo.toml there. There's no Cargo.lock.

This is on a fresh and updated Ubuntu 20.04 VM. I'm pretty sure this is the list of commands that got me to where I am now (maybe some are redundant; maybe I've messed some up):

sudo apt install rustc
sudo snap install --classic rustup
rustup install stable
rustup default stable
sudo apt install rust-src
rustup component add rust-src
sudo apt install gcc-avr
sudo apt install avr-libc
sudo apt install binutils-avr
sudo apt install avrdude
sudo apt install arduino-core
sudo apt install git

cd ~
git clone https://github.com/avr-rust/blink
cd blink

rustup override set nightly

export AVR_CPU_FREQUENCY_HZ=16000000
cargo build -Z build-std=core --target avr-atmega328p.json --release

Any ideas? Are there any good references out there to basically just step through what I'm trying to accomplish (the true hello-world embedded bring-up example)? I've kind of just pieced together info from a few different places at this point.

Thanks!

JosipHaboic, 2021-07-27:
@mjs717 The AVR in Rust is new, it does not goes smooth, you do not have that file because you cannot compile it, you need to set other version of Rust read this: https://book.avr-rust.com/001-introduction.html

stappersg, 2022-06-08:
Closing this.

@mjs717 please do use `rustup` from https://rustup.rs/ upon your retry.  Do report success here  and possible errors in a fresh issue.

closed 2022-06-08 by stappersg