avr-rust/blink #37: couldn't allocate input reg for constrain 'w'

str0yd, 2021-08-31:
I get this error compiling the project.
The whole error message:
`PS C:\Users\woppe\Documents\Rust\blink> cargo build -Z build-std=core --target avr-atmega328p.json --release
   Compiling compiler_builtins v0.1.49
warning: target json file contains unused fields: no-compiler-rt

   Compiling avr-config v1.0.0
   Compiling avr-config v2.0.1
   Compiling avr-std-stub v1.0.2
   Compiling avr_delay v0.3.1
warning: `avr-config` (lib) generated 1 warning (1 duplicate)
warning: `avr-std-stub` (lib) generated 1 warning (1 duplicate)
warning: `avr-config` (lib) generated 1 warning (1 duplicate)
   Compiling ruduino v0.3.2
error: couldn't allocate input reg for constraint 'w'
  --> C:\Users\woppe\.cargo\registry\src\github.com-1ecc6299db9ec823\avr_delay-0.3.1\src\lib.rs:34:17
   |
34 |         unsafe {llvm_asm!("1: sbiw $0,1
   |                 ^

warning: `avr_delay` (lib) generated 1 warning (1 duplicate)
error: could not compile `avr_delay` due to previous error; 1 warning emitted
warning: build failed, waiting for other jobs to finish...
LLVM ERROR: Not supported instr: <MCInst 312 <MCOperand Reg:1> <MCOperand Imm:13> <MCOperand Reg:41>>
warning: `compiler_builtins` (lib) generated 1 warning
warning: `ruduino` (lib) generated 1 warning (1 duplicate)
error: build failed`

sawyeke, 2021-10-09:
I am new to rust and just got the same issue. With some googlefoo I located [this issue](https://github.com/avr-rust/delay/issues/10). The comments from that one are that there's some processor specific work that needs to be done in order for the new asm! macro to work properly. Looks like a manpower issue more than anything else right now. Most of the recent work on the avr-rust project's been by @dylanmckay, and from the looks of it he's working more on other projects right now (which I totally understand, especially if he's mostly been doing this in his spare time).

axiomatic-aardvark, 2021-11-11:
I get the same issue :/

kvisscher, 2021-12-25:
Probably why this commit was made: https://github.com/avr-rust/blink/commit/9e9793050e61be90aaf1b403425915021caf6f50 to pin the version to a nightly that does still work