avr-rust/blink #38: cannot find macro `llvm_asm` in this scope

zag2art, 2022-01-30:
hi, getting this during compiling
```
cargo build -Z build-std=core --target avr-atmega328p.json --release
...

   Compiling avr_delay v0.3.1
error: cannot find macro `llvm_asm` in this scope
  --> /Users/zag2art/.cargo/registry/src/github.com-1ecc6299db9ec823/avr_delay-0.3.1/src/lib.rs:34:17
   |
34 |         unsafe {llvm_asm!("1: sbiw $0,1
   |                 ^^^^^^^^

error: cannot find macro `llvm_asm` in this scope
  --> /Users/zag2art/.cargo/registry/src/github.com-1ecc6299db9ec823/avr_delay-0.3.1/src/lib.rs:41:13
   |
41 |     unsafe {llvm_asm!("1: sbiw $0,1
   |             ^^^^^^^^

```

is that an avr_delay issue or this blink project?

Note: compiling this on Mac folowing "The AVR-Rust Guidebook"

stappersg, 2022-01-30:
> 
> Note: compiling this on Mac folowing "The AVR-Rust Guidebook"
> 

And what is the URL of "The AVR-Rust Guidebook"?

zag2art, 2022-01-30:
https://book.avr-rust.com/003.3-example-building-blink.html

stappersg, 2022-01-30:
It is https://github.com/avr-rust/delay/issues/10

stappersg, 2022-01-30:
For what it is worth
```text
    ...
   Compiling avr-config v1.0.0
   Compiling avr-config v2.0.1
   Compiling avr-std-stub v1.0.2
   Compiling avr_delay v0.3.1
   Compiling blink v0.1.0 (/usr/src/rust/avr/blink)
    Finished release [optimized] target(s) in 33.68s
stappers@paddy:/usr/src/rust/avr/blink
$ cargo build -Z build-std=core --target avr-atmega328p.json --release
    Finished release [optimized] target(s) in 0.11s
stappers@paddy:/usr/src/rust/avr/blink
$ history | tail -n 6
 2070  rustup override set nightly-2021-01-07
 2071  cargo build -Z build-std=core --target avr-atmega328p.json --release
 2072  rustup component add rust-src
 2073  cargo build -Z build-std=core --target avr-atmega328p.json --release
 2074  cargo build -Z build-std=core --target avr-atmega328p.json --release
 2075  history | tail -n 6
stappers@paddy:/usr/src/rust/avr/blink
$ 
```

zag2art, 2022-01-31:
Thanks
```
rustup override set nightly-2021-01-07
rustup component add rust-src
cargo build -Z build-std=core --target avr-atmega328p.json --release
```
Looks like it works only with nightly-2021-01-07 release.

stappersg, 2022-01-31:
Re-opened.
To prevent new reports of it.

Those who are also bitten by this issue (can|should) put updates here.

ShaneEverittM, 2022-05-06:
So currently the only solution is to use an old nightly compiler? Is the ultimate goal to change the implementation to not depend on `llvm_asm!`?

stappersg, 2022-05-06:
The ultimate goal is a good live.

> So currently the only solution is to use an old nightly compiler? Is the ultimate goal to change the implementation to not depend on `llvm_asm!`?

Yes.  Indeed.

How long _currently_ will exist is unknown.  ( It is unknown to me how the _LLVM project_ moves forward )
Known there are _workaround projects_ such as https://www.reddit.com/r/rust/comments/tvtxpv/custom_rust_toolchain_for_avr/

stappersg, 2022-05-21:
OKay, _LLVM project moved forward_, the [blocking bug is closed](https://github.com/rust-lang/compiler-builtins/issues/400#issuecomment-1122211690).

@mbuesch did [Fix build: Use the new rust asm! available in rustc 1.63.0-nightly](https://github.com/avr-rust/delay/pull/18) and [advised me](https://github.com/avr-rust/delay/issues/19#issuecomment-1133589844) on releasing `avr_delay` crate.

stappersg, 2022-06-08:
Insteadof crates at https://crates.io are "crates" in git repositories used.

The error `cannot find macro llvm_asm in this scope` is solved.

closed 2022-06-08 by stappersg