avr-rust/blink #41: could not load config key `env.AVR_CPU_FREQUENCY_HZ`

stappersg, 2022-06-06:
Since #40 
```text
stappers@myhost:~/src/rust/RustAVR/blink
$ cargo version
cargo 1.63.0-nightly (38472bc19 2022-05-31)
stappers@myhost:~/src/rust/RustAVR/blink
$ cargo build
error: error in /home/stappers/src/rust/RustAVR/blink/.cargo/config.toml: could not load config key `env.AVR_CPU_FREQUENCY_HZ`

Caused by:
  data did not match any variant of untagged enum EnvConfigValueInner
stappers@myhost:~/src/rust/RustAVR/blink
$ 
```

stappersg, 2022-06-06:
According https://doc.rust-lang.org/nightly/cargo/reference/config.html#env should it work.

Does it work for you?   (So that I know _it is just me_)

chinoto, 2022-06-07:
Same problem, same version. If I comment it out, I get an error compiling.
```
❯ cargo build
   Compiling core v0.0.0 (/home/damian/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
warning: target json file contains unused fields: no-compiler-rt

   Compiling rustc-std-workspace-core v1.99.0 (/home/damian/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
error: ran out of registers during register allocation

   Compiling compiler_builtins v0.1.73
warning: `rustc-std-workspace-core` (lib) generated 1 warning (1 duplicate)
warning: `core` (lib) generated 1 warning
error: could not compile `core` due to previous error; 1 warning emitted
warning: build failed, waiting for other jobs to finish...
warning: `compiler_builtins` (lib) generated 1 warning (1 duplicate)
```
core compiles if I use release mode (maybe that has always been required?), but then avr_delay complains that llvm_asm macro and feature doesn't exist, which sort of makes sense since there has been talk of deprecating it, though shouldn't it still exist for quite awhile?

chinoto, 2022-06-07:
I updated to the latest nightly yesterday and the llvm_asm issue went away, but I didn't have enough time to say anything at the time. Do nightly features occasionally disappear temporarily or was that sheer coincidence?

rockboynton, 2022-06-26:
Adding quotes around the value seemed to fix the issue
```toml
[env]
AVR_CPU_FREQUENCY_HZ = "16000000"
```

PR #45 (2022-06-28): Fix CPU frequency env var in config.toml
Fixes #41
commit d95f1a55 (2022-06-28): Fix CPU frequency env var in config.toml

Fixes #41
closed 2022-06-28 by stappersg