avr-rust/blink #46: LLVM ERROR: Expected a constant shift amount

saki-osive, 2023-05-14:
I'm trying to build this for an Arduino UNO. The spec sheet of the device can be found here: https://docs.arduino.cc/resources/datasheets/A000066-datasheet.pdf

I run the following commands before building:

```
rustup override set nightly
```
 and 

```
export AVR_CPU_FREQUENCY_HZ=16000000
```
to set the desired environment.

Then I run the following command to build for the board:

```
cargo build -Z build-std=core --target avr-atmega328p.json --release
```

The `avr-atmega328p.json` file is included at the root level.

However I'm faced with the following error:

```
..
   Compiling adler v1.0.2
   Compiling rustc-demangle v0.1.21
LLVM ERROR: Expected a constant shift amount!
error: could not compile `compiler_builtins` (lib)
warning: build failed, waiting for other jobs to finish...

```

Has anyone faced this before?

Patryk27, 2023-06-03:
Hi, yes, it's a known issue - https://github.com/rust-lang/rust/issues/112140; I'm working on a fix 🙂

b-camacho, 2023-08-20:
Thanks for fixing this! The fix already landed in nightly, in the mean time
`rustup toolchain install nightly` fixes this.

stappersg, 2025-06-21:
close

closed 2025-06-21 by stappersg